OpenI6X/opentx #381: Use nano ELRS rx (e.g. EP1/EP2) as internal ELRS module

wimalopaan, 2023-12-06:
The EP1/EP2 nano ELRS receivers can be flashed as TX-module. This can be doen for e.g. TX16s. Here a full-duplex connection to the module is possible.

This could replace the internal module of the ix6.

Maybe it would be possible to attach the RX as module via half-duplex. This would surely require a little hw-mod for the module (diode or bjt/fet). The RX could be mounted onto a adapter pcb with the half-duplex mod, and this could replace the internal module.

wimalopaan, 2023-12-06:
> It can be done half-duplex as it is now using TX2 pin, or even full duplex by also utilizing RX2 on PA15. This is available as external module.

Aside from configuring the ELRS RX itself, do I have to change something in openix6 to choose full- or half-duplex? Or is it recognized atomatically?

> 
> What do you mean by replacing internal module, ie: to add internal CRSF mode and leave external modes as they are?

Yes, I meant replacing the internal module. But since it is soldered and uses SPI this looks impossible.

ajjjjjjjj, 2023-12-06:
Currently CRSF is always using single pin, but full duplex mode and additional option in menu can be added. 
Additional benefit would be possibility to enable DMA fifo on RX part.

Current driver using `USART2` and `DMA Channel 4` for TX is in:
`radio/src/targets/flysky/telemetry_driver.cpp`

wimalopaan, 2023-12-07:
Actually I have not luck with EP2 as external module. 

Configured elrs to use tx pin same as rx pin. tried pin 3 and pin 1 for both. Tried elrs 3.0.1 and elrs 3.3.1.

wimalopaan, 2023-12-07:
I looks like the EP1/EP2 can't invert their RX/TX lines. 
I verified that inverting these line by HW-inverter and then tied together with pullup-diode, the EP2 can communicate with the radio. 
To simplify the wiring it would be best to make the TX2 uninverted(!) half-duplex as option. Then a simple pull-down diode would suffice for attaching a EP1/2 as tx-module.

Question: would it be possible to make TX2 uninverted half-duplex?

PR #383 (2023-12-08): Optionaly uninvert CRSF pin.
Closes: #381 
Closes: #382