OpenI6X/opentx #386: Porting to the Flysky i6s

Denniso2, 2024-01-19:
Is it possible to port this build to the flysky i6s? They two controllers use the exact same processor but the i6s has a touchscreen instead.
I have attached below some images of the internals.

[image]
[image]

ajjjjjjjj, 2024-01-19:
Yes. If you are a programmer you can contact me on discord regarding details.

Denniso2, 2024-01-19:
I am but I do not have experience with opentx. I will hit you up on discord.

wimalopaan, 2024-10-14:
Any progress here?

ajjjjjjjj, 2024-10-21:
None AFAIK, but i'm willing to help even when non working PR will be created with added i2c touch driver. 
Buttons can be remapped to touch regions.