OpenI6X/opentx #387: editing GVARS - system reboot

didinski, 2024-01-28:
Hello

when I set GVARS in Flight Modes the system sometimes makes a reboot? and an exclamation mark appears in the upper right corner.
I've tried factory reset but its still present. I'm not sure what causes the problem. But its always when I edit GVARS > #6.
GVARS 1-5 seem to work as expected.

using 1.11.1

ajjjjjjjj, 2024-06-08:
Can you add detailed steps and/or video of issue reproduction?

didinski, 2024-06-21:
just saw your answer..

i cruicial point was to edit many gvars (min. 1-6) and then edit >5. as far as i remember.

i will have a look and come back

ajjjjjjjj, 2024-06-28:
Can you attach yout eeprom configuration here?

didinski, 2024-06-28:
its always when i edit the **Name** of **GVAR6**. checked it. 
tested with 1.11.1, 1.11.0 and 1.10.0

[EEPROM.zip](https://github.com/user-attachments/files/16036589/EEPROM.zip)

ajjjjjjjj, 2024-07-30:
I have tested it on your config, seems to be fixed. Can you try by yourself on https://github.com/OpenI6X/opentx/pull/411 ?

PR #411 (2024-09-16): GVars menu issue, (Backport: Fix #7601)
GVars menu issue, (Backport: Fix #7601)

closes: #387
closed 2024-09-16 by ajjjjjjjj