OpenI6X/opentx #390: Is variometer implemented using FlySky FS-CAT01 altimeter sensor?

gbesoain, 2024-03-15:
I can't find any reference of if this sensor is implemented as vario sensor too, like is on https://hackaday.io/project/175220-flysky-variometer/details for openTx or as is on FlyPlus

ajjjjjjjj, 2024-03-16:
Hi, it is not, but I'm happy to add it if someone want to spend time to integrate this solution.

gbesoain, 2024-03-20:
Hi and thanks for your reply!

I'll be glad to try to integrate and test this solution, but I can't find the way nor instructions on how to build it.... I'm being trying to build using OpenTX instructions, but they are a bit outdated regarding software versions (QT Studio for instance) and I didn't be able to build it on my Windows 11 PC.

ajjjjjjjj, 2024-03-20:
Have you tried solution in wiki?

ajjjjjjjj, 2024-03-20:
I don't think we are talking about the same. Have you tried this way?
https://github.com/OpenI6X/opentx/wiki/Development#compile

gbesoain, 2024-03-20:
> I don't think we are talking about the same. Have you tried this way? https://github.com/OpenI6X/opentx/wiki/Development#compile

I missed those instructions, will try and let you know back

gbesoain, 2024-03-20:
Build done, using docker as instructed.

Question... I've added the lua script under "radio/sdcard/mixes" directory. It's ok or should be added on another directory?

ajjjjjjjj, 2024-03-21:
Lua is not supported on i6X, you need to rewrite it to C.