OpenI6X/opentx #404: Delta mode - can not get correct servo direction when in delta mix

thunderstar5000, 2024-06-20:
Dear developers, 

I do not know, if this is a bug, a feature problem or a knowledge problem by me. Please let me shortly describe:

Transmitter is a FlySky FS-i6 with firmware openi6x, version 1.11.1

For a delta I created to channels with aileron (Ch1 + Ch5) and mixed both each with elevation.
(aileron is set to 20%, elevation to 80%). 

For correct movement, I had to inverse CH1 (aileron, right wing), whilst the other servo (aileron, left wing) was stayed normal.

Now, the problem is: I can NOT get the right wing (CH1) running into correct direction with elevation. As the servo of the right wing had to be inversed, this affected also the direction of the elevation. 

I tried different combination, for hours, but found no solution. 

So I think, maybe this is not possible at all, or maybe it is a feature missing or I am doing something very wrong. 

It would be nice, if you could either confirm this as a bug or feature miss, or give me a clue, what I am doing wrong.

Beside of this, I am very happy with the software!

Thank you for reading this and your feedback in advance. 

Best regards

Hans

didinski, 2024-06-22:
Hi Hans,

have you looked here: /watch?v=REjQ8PMRa0I ?

thunderstar5000, 2024-06-22:
Am Samstag, 22. Juni 2024, 07:35:14 CEST schrieb didinski:
Hi didinski,

as i do not find my own message on github, I am answering directly. 

In te meantime I found the solution: You will have to set th eweight in the 
mixer to a negative value (i.e. 70 to -70), then the direction will change.

This  is not very well documented, I found it in one of the several documents 
in the web. And it was only a single sentence, in which this issue was 
annotated, more in a subclause.

However, the issue is solved. If you find my bugreport on github somewhere, it 
can be safely closed.

Thanks for your help!

I am very pleased with opentx, although I can not use companion (my Flysky 
FS-i6x is not supported by companion), I am very happy of it. Majorley, 
because I can set everything on my plane, which I could not with the original 
software (some mixes did not work).

So, thanks again for all the work, you made someone happy!

Best regards

Hans
> Hi Hans,
> 
> have you looked here: /watch?v=REjQ8PMRa0I ?

closed 2024-06-22 by ajjjjjjjj