OpenI6X/opentx #412: Working telemetry without RSSI but with valid TRSS

Snekovo, 2024-07-30:
I have CROSSOVER Telemetry RX FR7012-A2 (AFHDS 2A)
does not send RSSI only TRSS

telemetry correctly displays A3 drive battery voltage

The problems are.. that when trying to set up an alarm for low voltage in the logical switches section, picking A3 from the list of variables and comparing to a set value of (for example) 8.4v using the a<x function, the switch is never made...as if A3 doesn't exist.. also possibly connected is that the telemetry screen shows the datza and bar graphs for A3 and TRSS correctly, but a message flashes at the bottom of the screen "NO DATA"...

OpenI6X expects an RSSI value to be received and considers the lack of that as "no valid telemetry"

Please correct that OpenI6X knows when it receives TRSS that it should also work with telemetry 

Thank you

ajjjjjjjj, 2024-07-31:
Can you show picture of telemetry screen with all sensors?

Snekovo, 2024-07-31:
> Can you show picture of telemetry screen with all sensors?

[image]
[image]
[image]

PR #413 (2024-09-21): Create fake RSSI from TRSS when former not present
Create fake RSSI from TRSS when former not present.

Fixes: #412
closed 2024-09-21 by ajjjjjjjj