OpenI6X/opentx #415: Is it possible to have USB-Serial for Telemetrt mirror?

Menginventor, 2024-08-07:
For recent version, the remote can interface via USB port in 2-mode, storage mode and Joytick mode. Is it possible to ad the third mode as USB Serial?
Normally STM32 USB can be configured as USB-Serial so we can use telemetry mirror to PC easily and neat without additional wiring.

ajjjjjjjj, 2024-08-09:
It is possible but requires some flash space. 
Theoretically it could be implemented as a on/off feature and used by disabling something else during compilation to fit into flash space.

Menginventor, 2024-08-21:
May I ask for feature request on this one?

robante15, 2024-10-15:
It would be interesting, as it would allow telemetry to be obtained from the PC, and applications to be developed to function more like a ground station.

ajjjjjjjj, 2024-10-15:
> It would be interesting, as it would allow telemetry to be obtained from the PC, and applications to be developed to function more like a ground station.

It is already possible using UART to USB converters.

PR #436 (2024-11-21): USB Serial modes
Enables USB Serial mode forced to Telemetry mirror, or debug output in DEBUG builds.

Closes #415
closed 2024-11-21 by ajjjjjjjj