OpenI6X/opentx #418: Windows 11 pro (AMD) A request for the USB device descriptor failed.

angelosne, 2024-09-15:
When I connect it as a usb controller it does not recognize it . I have seen two different reasons. See screenshots
With original firmware I didn't have a problem so I guess it should be on the FW side

Steps to reproduce the behavior:
1. Connect USB to pc
2. Select controller as USB Joystick

**Expected behavior**
Windows recognize the device as a controller

**Screenshots**
[image]
[image]

**General information**
 - OpenI6X version 1.11.1
 - Hardware , the usb cable that came with the controller

ajjjjjjjj, 2024-09-21:
Can you test on this build?
[opentx-i6x-1.11.1-en.zip](https://github.com/user-attachments/files/17083221/opentx-i6x-1.11.1-en.zip)

angelosne, 2024-09-23:
I have already upload this FW but I haven't test it on my pc yet. I will let you know when I will test it! Thank you

angelosne, 2024-09-29:
> @angelosne, can you test on this build? [opentx-i6x-1.11.1-en.zip](https://github.com/user-attachments/files/17083221/opentx-i6x-1.11.1-en.zip)

I just test it. Same thing happens.

ajjjjjjjj, 2024-10-07:
Please test on 1.12.0, it should be finally fixed.

angelosne, 2024-10-13:
Congrats!! It's fixed! Thank you very much!
Now I open the joystick settings on WIN 11  and I can see this 
[image]

Left joystick vertical = Z axis
Left joystick horizontal = X Rotation
Right joystick = Y Axis/Y Axis rectangle
All switches work from left to right 1-4
VR A = Slider

I do not see VR B change anything. Is something that I have to setup?

closed 2024-10-14 by ajjjjjjjj

angelosne, 2024-10-15:
Yes it is working. It was my mistake because I do not know how to set it up! 
All good now! Thank you