OpenI6X/opentx #446: External power control (PC13) is not set after power-on, only if enabling/disabling from model menu

wimalopaan, 2024-12-12:


ajjjjjjjj, 2024-12-12:
Do you know what revision made regression and can you share your eeprom data because i can't reproduce?

wimalopaan, 2024-12-12:
... and I made a full EEPROM reset, but that didn't solve the problem.

wimalopaan, 2024-12-13:
If the type of the ext. module is PPM, then the PC13 functions as expected. So, initializing the crsf-stuff the setting of this pin is missing. I tried to debug a bit, but could not localize the bug in a first attempt.

wimalopaan, 2024-12-14:
This is called if started with 'PPM' as ext module:

```
opentxInit
opentxStart(0)
setupPulses 0: 0->1
disable_no_pulses 0
intmoduleStop
init_no_pulses 0
intmoduleNoneStart
extmoduleTimerStart 18000 0
setupPulses 1: 0->2
disable_no_pulses 1
extmoduleStop
DisablePPsOut
init_ppm
extmodulePpmStart
```

This is called if started with 'CRSF'as ext module:

```
opentxInit
opentxStart(0)
setupPulses 1: 0->3
disable_no_pulses 1
extmoduleStop
DisablePPMOut
telemetryPortInit 400000
setupPulses 0: 0->1
disable_no_pulses 0
intmoduleStop
init_no_pulses 0
intmoduleNoneStart
extmoduleTimerStart 18000 0

```
In 'extmoduleTimerStart' the PC13 pin should be set if the 'state' argument is 'true', but that never happens.

wimalopaan, 2024-12-14:
Looks like the problem is here:

```
void init_no_pulses(uint32_t port) {
  TRACE("init_no_pulses %d", port);
  if (port == INTERNAL_MODULE) {
    intmoduleNoneStart();
  }
//   else {
    extmoduleTimerStart(18000, false);
//   }
}
```
Removing the comments seems to fix it.

ajjjjjjjj, 2024-12-14:
I remember this part, I was hoping to never touch it again 🙈

wimalopaan, 2024-12-14:
Looks like I having a problem with the EEProm. 
Using an official build, the eeprom-save-caret does not vanisch after eeprom reset.

ajjjjjjjj, 2024-12-14:
Auto save is disabled when usb is connected - maybe this? This is standard OpenTX logic, but i don't know the reason behind it, maybe needs to be changed.

wimalopaan, 2024-12-14:
No USB used in my tests

wimalopaan, 2024-12-14:
Can you reproduce that with my cmake?

wimalopaan, 2024-12-14:
I found that `setupPulses()` is permanently called ... looks like to root cause here.

wimalopaan, 2024-12-14:
I added further debug messages, but this part looks like a total mess. I'm not able to get through ... It's difficult to debug because `setupPulses()` is called permanently.

wimalopaan, 2024-12-14:
No I get `Unexpected Shutdown detected` calling `disable_no_pulses(port);` and an exclamation mark in the upper right corner of the LCD.

ajjjjjjjj, 2024-12-14:
Whole part needs porting current opentx/edgetx solution or investigation if ppm setup is done correctly

ajjjjjjjj, 2024-12-14:
And out of sudden someone just asked about PPM on discord... 
But it definitely needs improvements, maybe it will be not that hard.
I don't like "simplification" where PPM IN uses same timer as ext module, when we have spare TIM2, which break module toggle logic in effect.

wimalopaan, 2024-12-14:
> But it definitely needs improvements, maybe it will be not that hard.
> I don't like "simplification" where PPM IN uses same timer as ext module, when we have spare TIM2, which break module toggle logic in effect.

It would be very cool, if you could donate some time on that topic.

ajjjjjjjj, 2024-12-16:
I haven't tested that yet, but it seems that `EXTMODULE_TIMER` is configured in wrong place. Should be in `extmodulePpmStart()` and `intmoduleNoneStart()`, then logic with enabling else statement in `init_no_pulses` should not break anything.

wimalopaan, 2024-12-17:
> I haven't tested that yet, but it seems that `EXTMODULE_TIMER` is configured in wrong place. Should be in `extmodulePpmStart()` and `extmoduleTimerStart()`, then logic with enabling else statement in `init_no_pulses` should not break anything.

I really hope you'll get through that mess ...

ajjjjjjjj, 2024-12-18:
I found it but it needs mayor refactorization of this true mess.
Issue is external module timer never starts for OFF or CRSF mode, but should be always running. Thats also why watchdog heartbeat was broken in the first place. So another mistery almost solved.

wimalopaan, 2024-12-19:
> I found it but it needs mayor refactorization of this true mess.

Great!!!
I wasn't able to discover any logic in the flow of calls ...

PR #455 (2025-01-20): Pulses refactor
- Pulses refactor to match current OpenTX/EdgeTX logic.
- Properly enable/disable PPM in/out CC registers.
- Disable Trainer PPM IN interrupt when turned off,
- Cleanup EXTMODULE_TIMER_IRQHandler logic.
- Fix EXTMODULE power pin toggle. 
- Merge pulses_arm.h and pulses.h.
- Removed pulses_driver.cpp. 
- Cleanup

Closes #446
closed 2025-01-20 by ajjjjjjjj