OpenI6X/opentx #459: 2.0.0 - Logical Switches - Editing - Up/Down functionality stops working

markhen57, 2025-01-28:
**Describe the bug**
While editing Logical Switches *sometimes* the up/down buttons stop working after returning from Edit or Clear. In this case pressing Return twice will return cursor to the "page" selection field ("8/11") and up/down resumes working.

**To Reproduce**
I have been unable to identify a single sequence that will recreate this every time, while editing Logical Switches this has happened multiple times today.

**Expected behavior**
Up/down should work consistently.

**General information**
 - OpenI6X version (e.g. 2.0.0)
 - Hardware - DFPlayer

ajjjjjjjj, 2025-02-01:
Thanks for reporting!
Can you test this builds for fix + additional improvements described here: https://github.com/OpenI6X/opentx/pull/460 ? 
Firmware to test is here: https://app.circleci.com/pipelines/github/OpenI6X/opentx/1398/workflows/cb91d4c6-e6cf-4b97-80f5-a8f7150b3760/jobs/3529/artifacts

closed 2025-02-02 by ajjjjjjjj

markhen57, 2025-02-02:
Thank you! I have tested and it works as expected.
    
    
      Thank you for this incredible software port - the FS-i6X with Openi6X is 
      now my favorite radio (form factor, functionality etc.). Love it! Please 
      keep going! If you need/want other testers I am available any time!
    
    
      Cheers
    
    
      Mark
    
    
      
      
      
      -----Original Message-----From: JanekSent: Sat 
      1:34pm Feb 1 2025To: ***@***.***: [OpenI6X/opentx] 
      2.0.0 - Logical Switches - Editing - Up/Down functionality stops working 
      (Issue #459)
    
  

Thanks for reporting!
Can you test this builds for fix + additional improvements described here: #460 ?
Firmware to test is here: https://app.circleci.com/pipelines/github/OpenI6X/opentx/1398/workflows/cb91d4c6-e6cf-4b97-80f5-a8f7150b3760/jobs/3529/artifacts

?
[
{
"target": "https://github.com/OpenI6X/opentx/issues/459#issuecomment-2629123887",
"url": "https://github.com/OpenI6X/opentx/issues/459#issuecomment-2629123887",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

ajjjjjjjj, 2025-02-03:
Thanks a lot for this kind words!

PR #460 (2025-02-02): Fixed issue where returning from popup was blocking UP/DOWN navigation,
- Fixed issue where returning from popup was blocking UP/DOWN navigation,
- added shortcut to go directly to edit mode for empty LS, 
- removed POT navigation code,
- some definitions fixes for excluding not used codepath,

Closes #459