OpenI6X/opentx #461: Adapt CRSF implementation to TBS CRSF specs

wimalopaan, 2025-02-12:
This maybe more a question than a real bug:

The TBS specs for CRSF folder type parameters specifies, that the client should send the IDs of the contained subitems in order to speedup the menu display. As I'm actually away from the bench (no setup with LA at hand), I can't confirm myself if the OpenI6X implementation of this feature is conforming.

If anyone is testing please be aware that the is a ELRS flaw: https://github.com/ExpressLRS/ExpressLRS/issues/3121 , so that testing with actual ELRS is impossible. You need to use TBS gear or some other conforming implementation.

ajjjjjjjj, 2025-02-12:
Hey, i'm aware of that difference but because elrs implemented that wrong we have to stick to basic implementation iterating over all ids.
This is a bit hacked in OpenI6X - ids lower than folder id are skipped anyway.
Current implentation was tested by users of original TBS hardware with success.

ajjjjjjjj, 2025-03-31:
Implemented in ELRS as of https://github.com/ExpressLRS/ExpressLRS/pull/3123