OpenI6X/opentx #470: External Module Power Diagram - Wiki - P-MOSFET Drain/Source Swapped

markhen57, 2025-04-03:
**Describe the bug**
External Module Power circuit in the Wiki shows the P-MOSFET with Drain terminal connected to +V and Source terminal connected to load. 
In a normal P-MOSFET the intrinsic or body diode has its cathode connected to Source and anode connected to Drain. 
As shown in the Wiki the load will always receive power from the circuit - either through the MOSFET (when turned on) or through the body diode (with one Vf drop). 

**To Reproduce**
Build the circuit and see :-)

**Expected behavior**
The correct connection of the P-MOSFET is to have the Source connected to V+ and the Drain connected to the load. Then the body diode is reverse biased at all times.

**Screenshots**
P-MOSFET should be oriented as below in the Wiki diagram:

[image]

**General information**
 - Wiki as of April 3, 2025

ajjjjjjjj, 2025-04-05:
@marco-garcia , can you fix this?

marco-garcia, 2025-04-05:
@markhen57 is correct, I've made a mistake in the drawing

I can't fix it right now, but by tomorrow I'll post a corrected diagram

My apologies for the oversight

marco-garcia, 2025-04-07:
here we go @ajjjjjjjj  :

[image]

ajjjjjjjj, 2025-04-07:
Great!

closed 2025-04-07 by ajjjjjjjj

markhen57, 2025-04-09:
Thanks gentlemen!