OpenI6X/opentx #484: [v2.1.0] Firmware doesn't respect CUSTOM SET Failsafe values for VRA/VRB switches - abnormal values

zigapovhe, 2025-08-12:
**Describe the bug**
When transmitter is turned off, receiver starts outputing abnormal values on VRA/VRB channels. Changing CH Failsafe setting does change the numbers but are still abnormal. 

Without failsafe mode, when reading values of VRA/VRB:
MIN: 988
MAX: 2012

Global functions are disabled, Flight modes not set, Inputs are default, Mixers also default, Outputs - default, Curves - not set, Logical switches - not set, Special functions - not set.

If you need any more info, i would be happy to provide it. 

**To Reproduce**
In the SETUP menu, Failsafe is set to CUSTOM. Below is the description of custom failsafe settings for CH6

FAILSAFE SETTINGS: 
- HOLD
    - CH6: NONE  
    - ACTUAL RECEIVER VALUE IN FAILSAFE: 34780(If last value was min) 33756 (if last value was max) - Difference: 1024
    
    - CH6: 0
    - ACTUAL RECEIVER VALUE IN FAILSAFE: 9692 - always (actual last value doesn’t matter)
     
    - CH6: -100
    - ACTUAL RECEIVER VALUE IN FAILSAFE: 10204
     
    - CH6: 100
    - ACTUAL RECEIVER VALUE IN FAILSAFE: 10204
     
    - CH6: HOLD
    - VALUE BEFORE FAILSAFE: 2012
    - ACTUAL RECEIVER VALUE IN FAILSAFE: 34780
    - VALUE BEFORE FAILSAFE: 988
    - ACTUAL RECEIVER VALUE IN FAILSAFE: 33756

**Expected behavior**
Expected behavior would be for receiver(in failsafe mode) to output value between 988 - 2012. 

**General information**
 - OpenI6X version: v2.1.0 
 - Hardware: FlySky FS-I6X with FS-IA10B receiver
 - Protocol: IBUS

zigapovhe, 2025-08-13:
> This is not an issue or bug. The transmitter(Openi6X) has zero influence over the receiver when it is not powered on, therefore the failsafes you are referring to in this issue should be implemented by your receiver or flight controller if applicable. This is because the receiver has its own firmware which is separate from Openi6X.

Ah thanks, I was wondering about that yes. I don't know how everything works, but is it possible that transmitter sends wrong data to the receiver when saving failsafe settings? It is weird that settings on all other channels works fine except for VRA and VRB (potentiometers). I guess its a bug in the receiver firmware then.

BryanSouza91, 2025-08-13:
Glad you found the culprit.

zigapovhe, 2025-08-13:
Thanks will check it out.
I'm closing this issue as high-bit flags were indeed the problem. Everything works as expected now. Thank you for your time and info.